Cloud-CNC/cura-wasm #3: nable to get it to work (worker.js problem)

RaulLazaro, 2020-09-29:
Hello,

I am trying to test this package as it looks very interesting but I am unable to get it to work.

First I tried a project with **Angular**, create a simple component very similar to your example.

```
import { Component, OnInit } from '@angular/core';
import {CuraWASM} from 'cura-wasm/dist/es';

@Component({
  selector: 'app-root',
  templateUrl: './app.component.html',
  styleUrls: ['./app.component.scss']
})
export class AppComponent implements OnInit {
  //Create a slicer
  slicer = new CuraWASM({
    definition: 'ultimaker2',
    verbose: true
  });
  stl: any;
  percent: number = 0;
  elapsed: Date = new Date(0);
  gcode: ArrayBuffer;

  ngOnInit() {
    //Add progress handler
    this.slicer.on('progress', percent => {
      this.percent = percent;
    });
  }

  handleFileInput(files: FileList) {
    if (files && files.length > 0) {
      this.stl = files.item(0);
    } else {
      this.stl = null;
    }
  }

  async slice() {
    const start = Date.now();
    this.gcode = await this.slicer.slice(this.stl)
    const end = Date.now();

    //Calculate elapsed time
    this.elapsed = new Date(end - start);
  }
}
```

But when clicking on slice I get this error.
[image]

In a second attempt, I copied your code from the **demo folder** and use webpack to bundle the js, but the same error occurred.

I don't know if any extra configuration is necessary, perhaps from webpack, or there is something wrong with the package.

lats339, 2020-09-29:
I am not really the person to ask for this, contact wakeful cloud and see
what he says

Wakeful-Cloud, 2020-09-29:
@RaulLazaro May I ask what version you're using? I apologize for the example not working.

RaulLazaro, 2020-09-29:
I started with version v1.1.1-beta.0 but today I am testing with the last one, v1.1.1

Wakeful-Cloud, 2020-09-29:
@RaulLazaro I believe the problem is with serving `worker.js`. Can you verify if your dev server is serving it at `http://localhost:4201/worker.js` (It should be a minified JS file)?

RaulLazaro, 2020-09-29:
it is not being served. But I don't know how to configure it because the worker.js file is the one inside node_modules/cura_wasm/dist/es ?

Wakeful-Cloud, 2020-09-29:
@RaulLazaro Yes, it's located at `node_modules\cura-wasm\dist\es\worker.js`. I'm afraid this might be a bug with `/index.js` being hard-coded into `cura-wasm` preventing all but the demo from working.

Wakeful-Cloud, 2020-09-29:
@RaulLazaro Could you post your webpack + dev-server configuration?

RaulLazaro, 2020-09-29:
I'm using the default ones, I don't have much experience with webpack.

I simply run `npx webpack` to bundle `src/index.js` to `dist/main.js`

And then serve with npm script:
```
"scripts": {
  "start": "webpack-dev-server"
}
```

Wakeful-Cloud, 2020-09-29:
@RaulLazaro Can you check if the worker.js is served under the `dist` directory? (`http://localhost:4201/dist/worker.js`)

closed 2020-10-03 by Wakeful-Cloud

Wakeful-Cloud, 2020-10-03:
@RaulLazaro cura-wasm@1.2.1 is working for me with Angular.

I did notice a bug in your implementation: `this.stl` was a reference to a File, not an ArrayBuffer. The fix is to call `this.stl.arrayBuffer()` which returns a promise that resolves to an ArrayBuffer.

Please tell me if anything else does not work and thanks for filing this issue.

RaulLazaro, 2020-10-03:
It works great!! Thanks for your time.

PR #4 (2020-10-03): Webpack fix
Fixes #3