Cloud-CNC/cura-wasm #5: Every thing besides the wasm under MIT licence.

mmiscool, 2020-10-25:
I am interested in this having an MIT licence to access the api. In a traditional node environment the line is clear as you use a compiled binary and the shell command line interface. I am having trouble finding that line here.

Wakeful-Cloud, 2020-10-25:
@mmiscool The equivalent line would be [Worker.ts (Line 152)](https://github.com/Cloud-CNC/cura-wasm/blob/master/src/worker.ts#L152). Cura WASM doesn't start a separate process in the browser nor on NodeJS, it instead uses workers provided by [Threads.JS](https://threads.js.org) in both environments.

The most I would be able to do is MIT license everything except CuraEngine.js which contains the actual WASM (Base 64 encoded). If that would help your license compliance, I'd be happy to relicense everything except that file (As MIT).

mmiscool, 2020-10-25:
That could be beneficial. Thanks!

Wakeful-Cloud, 2020-10-26:
@mmiscool Closed in V1.3.2

closed 2020-10-26 by Wakeful-Cloud