Cloud-CNC/cura-wasm #6: Difficulty building the demo to a folder that can be placed on a plain web server

mmiscool, 2020-10-27:
Hello,

I have been trying to figure out how to output a directory of the demo that I can just copy to a web server and use with out having to use the nollup server thing. I seem to be getting stuck. Not really familiar with rollup and nollup. 

Any suggestions?

Wakeful-Cloud, 2020-10-27:
I apologize for not making this very clear but in the future, please ask for help in the [official Spectrum chat](https://spectrum.chat/cloud-cnc/cura-wasm).

I **strongly** suggest not running the demo for anything other than its intended purpose because it's only meant to be used with Nollup and no guarantee are made that this patch will work in the future. If you still want to do this, replace the **demo's** Rollup config (`./demo/rollup.config.js`) with the below code. Then run `npx rollup -c ./demo/rollup.config.js` from the **project's root directory**. Once done, copy and paste the files from the output (`./demo/public`) and the `./demo/index.html` into your web server's public directory (Or equivalent).
```Javascript
/**
 * @fileoverview Rollup demo config
 */

//Imports
import {terser} from 'rollup-plugin-terser';
import commonjs from '@rollup/plugin-commonjs';
import json from '@rollup/plugin-json';
import resolve from '@rollup/plugin-node-resolve';
import threads from 'rollup-plugin-threads';
import typescript from 'rollup-plugin-typescript2';

//Export
module.exports = {
  input: [
    'demo/index.js'
  ],
  output: [
    {
      dir: 'demo/public',
      format: 'cjs'
    }
  ],
  plugins: [
    resolve({
      browser: true,
      preferBuiltins: false,
      //Force Rollup to use the CJS version of Threads JS
      mainFields: [
        'main'
      ]
    }),
    json(),
    commonjs(),
    typescript(),
    threads({
      include: '**/worker.ts',
      plugins: [
        resolve({
          browser: true,
          preferBuiltins: false
        }),
        json(),
        commonjs(),
        typescript()
      ]
    }),
    terser()
  ],
  onwarn: (warning, warn) =>
  {
    //Hide eval warning from Emscripten
    if (warning.code != 'EVAL' && warning.code != 'THIS_IS_UNDEFINED')
    {
      warn(warning);
    }
  }
};
```

closed 2020-10-27 by Wakeful-Cloud

mmiscool, 2020-10-27:
Thanks. I will hop on that link you provided going forward.

Wakeful-Cloud, 2020-10-27:
@mmiscool Usually sometime between UTC-6 and UTC-8