Cloud-CNC/cura-wasm #7: RFC: Reduction of bundle size and separation of printer definitions

Wakeful-Cloud, 2020-10-31:
# RFC: Reduction of bundle size and separation of printer definitions

## Problem
The package bundle size of Cura WASM is too high. Package consumers substantially increase their own code bundle sizes simply by including Cura WASM which causes end users to suffer as a result.

## Duration
November 6th, 2020 (11/6/2020)
##### *Note that the Cloud CNC core development team reserves the right to extend the duration.*

## Proposers

* [Wakeful-Cloud](https://github.com/wakeful-cloud) - Propositions A, C
* [InventInside (p12)](https://github.com/inventinside) - Proposition B

## Detail
Cura WASM’s production bundled size is 4.05 MB (CommonJS version) or 4.04 MB (ES6+ version) as of writing. The package is bundled with 309 3D printer definitions (As of writing) taking up approximately 1.24 MB. The compiled Cura Engine Web Assembly is “inlined” inside of [CuraEngine.js](https://github.com/cloud-cnc/cura-wasm/tree/master/src/CuraEngine.js) using base 64 encoding.

If all proposals are adopted, the size of Cura WASM (Excluding 3D printer definitions) will shrink from 4.05 MB to 2.03 MB (CommonJS version; 49.88% decrease) or from 4.04 MB to 2.01 MB (ES6+ version; 50.25% decrease).

## Proposals
### Proposal A
<ins>Implementing size-optimizing GCC flags when building Cura Engine</ins> will produce substantially smaller builds with **NO IMPACT TO CURA WASM’S FUNCTIONALITY**. The performance can only be evaluated once this proposal had been adopted; if performance is deemed unacceptable, this proposal will be automatically rejected (Retrospectively). The GCC flags will be “`-Os`”. If size-optimizing GCC flags are used, [CuraEngine.js](https://github.com/cloud-cnc/cura-wasm/tree/master/src/CuraEngine.js) will shrink from 1.78 MB to 1.17 MB (34.27% decrease). The overall package bundle size will also shrink from 4.05 MB to 3.46 MB (CommonJS version; 14.57% decrease) or 4.04 MB to 3.44 MB (ES6+ version; 14.85% decrease).

### Proposal B
The <ins>elimination of printer definitions</ins> will reduce the package bundle size from 4.05 MB to 2.81 MB (CommonJS version; 30.62% decrease) or from 4.04 MB to 2.80 MB (ES6+ version; 30.69% decrease). Note that the user will still have to supply printer definitions. 3D printer definitions will be moved to a new GitHub repository and NPM package. The name of this package will be “`cura-wasm-definitions`”. Package consumers will then be responsible for supplying Cura WASM with the appropriate 3D printer definitions. This will cause **BREAKING CHANGES TO CURA WASM’S API**. Instead of specifying what 3D printer definition to use to Cura WASM, this same information will be implied by what 3D printer definition is supplied to Cura WASM.

### Proposal C
<ins>Splitting Cura WASM’s Cura Engine Web Assembly file</ins> will reduce the package size simply by avoiding the overhead associated with base-64 encoding the Web Assembly file. However, splitting the Web Assembly file will likely cause **BREAKING CHANGES** due to Emscripten’s proprietary file loading system. The size of [CuraEngine.js](https://github.com/cloud-cnc/cura-wasm/tree/master/src/CuraEngine.js) will shrink from 1.78 MB to 1.36 MB (Also including the new, separate Web Assembly file; 23.60% decrease). The overall package bundle size will also shrink from 4.05 MB to 3.63 MB (CommonJS version; 10.37% decrease) or 4.04 MB to 3.62 MB (ES6+ version; 10.40% decrease).

## Votes
Proposal A (>50% to pass): 0/0
Proposal B (70% to pass): 0/0
Proposal C (70% to pass): 0/0
##### *Note that the Cloud CNC core development team reserves the right to modify these voting thresholds to best suit the total number of voters.*

## Voter Policy
**Anyone informed on all of the above policies and their implications is welcome to vote.** Votes can be cast by commenting in the below format:
```Markdown
Official Vote
Proposition A: adopt/reject
Proposition B: adopt/reject
Proposition C: adopt/reject
```

## Resolution
Proposal A: TBD
Proposal B: TBD
Proposal C: TBD

Wakeful-Cloud, 2020-10-31:
Official Vote
Proposition A: adopt
Proposition B: adopt
Proposition C: reject

inventinside, 2020-11-02:
Official Vote
Proposition A: adopt
Proposition B: adopt
Proposition C: reject

mmiscool, 2020-11-07:
B

Wakeful-Cloud, 2020-11-07:
@mmiscool I'm not sure what that means. Does that mean you only want to adopt proposition B? Please use the template for voting. I'll leave this RFC open until tomorrow (If you don't change your post or reply with more details by then, your vote will not be counted).

Wakeful-Cloud, 2020-11-09:
Voting has closed - thanks to everyone who participated! Propositions A and B will be implemented in the coming days.

closed 2020-11-09 by Wakeful-Cloud

Wakeful-Cloud, 2020-11-16:
@inventinside @mmiscool

**Proposal A Rejection**
Proposal A has been retrospectively rejected due to unacceptable performance. The below chart shows the proposed flag (`-Os`) causing otherwise identical slices to take over 2 times as long. However, the flag `-O2` has been adopted due to negligible performance loss and minor size reductions (Over the existing `-O3`) FYI: each categorical value was averaged from 6 individual trials.

[image]