Cloud-CNC/cura-wasm #8: Overrides don't seem to apply

brunobasto, 2020-11-25:
Hello! Nice project! I'm experimenting with it and I can't seem to make the overrides work.

```js
const slicer = new CuraWASM({
    definition: 'creality_ender3',
    overrides: [{
        key: 'infill_sparse_density',
        value: '20'
    },
    {
        // scope: 'e0', this makes CuraWASM exit before even slicing
        key: 'layer_height',
        value: '0.2'
    },
    {
        key: 'infill_pattern',
        value: 'cubic'
    },
    {
        key: 'speed_print',
        value: '50'
    }
    ],
    transfer: false,
    verbose: true,
});
```

The generated gcode ends up ignoring my settings:

```gcode
;FLAVOR:Marlin
;TIME:6666
;Filament used: 0m
;Layer height: 0.1
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06

;Generated with Cura_SteamEngine master
```

As you can see, the layer height used as 0.1 and not 0.2 as I wanted. Also, it's weird that the `Filament used` is 0m, isn't it? I suppose it has to do with the positions. MINX, MINZ, etc look strange.

Here's what the same STL sliced with Cura  4.7.1 looks like:

```gcode
;FLAVOR:Marlin
;TIME:6765
;Filament used: 3.82669m
;Layer height: 0.2
;MINX:80.626
;MINY:98.406
;MINZ:0.2
;MAXX:147.3
;MAXY:136.595
;MAXZ:48
;Generated with Cura_SteamEngine 4.7.1
```

Wakeful-Cloud, 2020-11-25:
Thanks for reporting this. Can you post the output from Cura WASM? Also what version of Cura WASM are you running?

brunobasto, 2020-11-25:
Sure. The version is `"cura-wasm": "^1.3.3",` and here goes the log:

```
Initialized worker!
Added definitions!

Progress: 2%
Progress: 2%
Progress: 3%
Progress: 3%
Progress: 4%
Progress: 4%
Progress: 5%
Progress: 5%
Progress: 6%
Progress: 6%
Progress: 7%
Progress: 7%
Progress: 8%
Progress: 8%
Progress: 9%
Progress: 9%
Progress: 10%
Progress: 10%
Progress: 11%
Progress: 11%
Progress: 12%
Progress: 12%
Progress: 13%
Progress: 13%
Progress: 14%
Progress: 14%
Progress: 15%
Progress: 15%
Progress: 16%
Progress: 16%
Progress: 17%
Progress: 17%
Progress: 18%
Progress: 18%
Progress: 19%
Progress: 19%
Progress: 20%
Progress: 20%
Progress: 21%
Progress: 21%
Progress: 22%
Progress: 22%
Progress: 23%
Progress: 23%
Progress: 24%
Progress: 24%
Progress: 25%
Progress: 25%
Progress: 26%
Progress: 26%
Progress: 27%
Progress: 27%
Progress: 28%
Progress: 28%
Progress: 46%
Progress: 46%
Progress: 47%
Progress: 47%
Progress: 48%
Progress: 48%
Progress: 49%
Progress: 49%
Progress: 50%
Progress: 50%
Progress: 51%
Progress: 51%
Progress: 52%
Progress: 52%
Progress: 53%
Progress: 53%
Progress: 54%
Progress: 54%
Progress: 55%
Progress: 55%
Progress: 56%
Progress: 56%
Progress: 56%
Progress: 56%
Progress: 57%
Progress: 57%
Progress: 59%
Progress: 59%
Progress: 60%
Progress: 60%
Progress: 61%
Progress: 61%
Progress: 62%
Progress: 62%
Progress: 63%
Progress: 63%
Progress: 64%
Progress: 64%
Progress: 65%
Progress: 65%
Progress: 66%
Progress: 66%
Progress: 67%
Progress: 67%
Progress: 68%
Progress: 68%
Progress: 69%
Progress: 69%
Progress: 70%
Progress: 70%
Progress: 71%
Progress: 71%
Progress: 72%
Progress: 72%
Progress: 73%
Progress: 73%
Progress: 74%
Progress: 74%
Progress: 75%
Progress: 75%
Progress: 76%
Progress: 76%
Progress: 77%
Progress: 77%
Progress: 78%
Progress: 78%
Progress: 79%
Progress: 79%
Progress: 80%
Progress: 80%
Progress: 81%
Progress: 81%
Progress: 82%
Progress: 82%
Progress: 83%
Progress: 83%
Progress: 84%
Progress: 84%
Progress: 85%
Progress: 85%
Progress: 86%
Progress: 86%
Progress: 87%
Progress: 87%
Progress: 88%
Progress: 88%
Progress: 89%
Progress: 89%
Progress: 90%
Progress: 90%
Progress: 91%
Progress: 91%
Progress: 92%
Progress: 92%
Progress: 93%
Progress: 93%
Progress: 94%
Progress: 94%
Progress: 95%
Progress: 95%
Progress: 96%
Progress: 96%
Progress: 97%
Progress: 97%
Progress: 98%
Progress: 98%
Progress: 99%
Progress: 99%
Progress: 100%
Progress: 100%
slice: 7.738s
```

Wakeful-Cloud, 2020-11-25:
Hmm, that's odd - usually the log shows more. Did you disable warnings and errors in the inspector? (For some reason, Ultimaker decided to write to stderr for a lot of non-errors and so emscripten logs a bunch of error messages even when they're not errors.) Here's what I see when I enable verbose logging:
[image]

I'm starting to suspect that `Couldn't find definition file with ID: ultimaker2_extruder_0` may be a problem (Because this package and the new dedicated printer definition package, [`cura-wasm-definitions`](https://github.com/cloud-cnc/cura-wasm-definitions) (For `cura-wasm` >= 1.4.0 only) don't include extruder definitions but do include printer definitions hence the error message). I wonder if this would lead to Cura WASM slicing for a weird `fdmextruder` hybrid definition that takes some of the settings of the `fdmextruder` and some from the specified definition (eg: Ender 3 or Ultimaker 2). If I'm correct, I'd imagine that it would produce extremely odd and possibly unusable GCODE which could explain some of the oddities you noticed in your generated GCODE.

brunobasto, 2020-11-25:
Humm. I should mention that I'm running it on node. Not on a browser. Node version is v15.3.0 also tested on v14.12.0. Maybe the logs on node aren't so verbose?

Wakeful-Cloud, 2020-11-25:
Oh, yes, that probably explains the logging. Don't worry about trying to get those logs then!

Wakeful-Cloud, 2020-11-25:
I'm currently working on adding extruder definitions and see if it produces better GCODE (My GCODE has similar issues).
```

brunobasto, 2020-11-25:
Awesome! Looking forward to hear about your progress on it.

Wakeful-Cloud, 2020-11-26:
Unfortunately, adding extruder definitions did not resolve this problem. I sliced Benchy with Cura WASM and Cura Engine (Version 4.7.1; packaged with Cura "desktop" but not through Cura "desktop") and while they didn't produce identical files, the commented-out header was identical (Minus version information):

**Cura WASM Header**
```gcode
;FLAVOR:Marlin
;TIME:6666
;Filament used: 0m
;Layer height: 0.1
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06

;Generated with Cura_SteamEngine master
```

**Cura Engine Header**
```gcode
;FLAVOR:Marlin
;TIME:6666
;Filament used: 0m
;Layer height: 0.1
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06

;Generated with Cura_SteamEngine 4.7.1
```

As far as fixing this, [InventInside (p12)](https://github.com/inventinside) and I have been discussing this on the [official Spectrum](https://spectrum.chat/cloud-cnc/cura-wasm) generating GCODE more true to Cura "desktop", you may want to check that out. I'm going to have to think about this some more but it's important that this gets fixed.

brunobasto, 2020-11-26:
Thank you very much for your work on this. I will check out the spectrum discussion. I'm not very familiar with the WASM stuff, but I'll try to learn more about it to see if I can be of any help to the project.

Wakeful-Cloud, 2020-11-27:
@brunobasto Thanks! I'm hypothesizing that I was on the right track with thinking that some kind of definition is missing from Cura WASM - except that extruder definitions aren't necessarily the problem, maybe the print profiles/printer variants (Or lack of) are the problem. After looking through the [Cura resource directory](https://github.com/Ultimaker/Cura/tree/master/resources), I noticed that there were [`quality`](https://github.com/Ultimaker/Cura/tree/master/resources/quality) and [`variants`](https://github.com/Ultimaker/Cura/tree/master/resources/variants) directories. Each of them contains quite a few `.cfg` files specific to different printers. Since you observed issues with Cura WASM ignoring layer heights, I wonder if some kind of hard-coded default profile (In Cura Engine) was overwriting your arguments. I'll have to keep researching.

Wakeful-Cloud, 2020-12-26:
After further examination, there was a bug (With Cura WASM) in generating the CLI arguments used to call Cura Engine. This was likely causing Cura WASM to crash when you specified `e0` for the `layer_height` scope. This has been fixed in `V1.4.3`.

I'm still investigating the *layer_height setting being ignored* and the *odd filament usage*.

Wakeful-Cloud, 2021-01-01:
Fixed in V1.5.1

For anyone curious, the problem was that the order of the CLI arguments matters to Cura Engine.

closed 2021-01-01 by Wakeful-Cloud