Cloud-CNC/cura-wasm #9: Types definition point to inexistent folder

brunobasto, 2020-11-30:
Hello! Me again :)

https://github.com/Cloud-CNC/cura-wasm/blob/master/package.json#L7

I couldn't find cura-wasm/dist/types inside my node_modules. VSCode gives me compilation errors.

Wakeful-Cloud, 2020-11-30:
Hmm that's odd. Are you still running V1.3.3?

brunobasto, 2020-11-30:
Yes. Just checked. It's still 1.3.3. For me, there's no `types` folder inside `dist`:

[image]

I noticed this yesterday because I'm migrating my little project to typescript (while it's still small 🤣 )

Wakeful-Cloud, 2020-12-01:
The types live inside of the `cjs` and `es` directories. Currently, the `package.json` points to the nonexistent `types` directory but this will be fixed very soon. In the meantime, you can change the `"types"` field in Cura WASM's package.json to `"types": "dist/es/index.d.ts",`.

Wakeful-Cloud, 2020-12-01:
Please update to version 1.4.2 - switched to named exports, separated definitions to separate package ([`cura-wasm-definitions`](https://github.com/cloud-cnc/cura-wasm-definitions)), and fixed the package.json issues. Apologies for the confusion and thanks for filing.

closed 2020-12-01 by Wakeful-Cloud

brunobasto, 2020-12-01:
Awesome. Thank you