Cloud-CNC/cura-wasm #12: Definition "fdmprinter" doesn't work

benzend, 2021-06-28:
I've tried using 'ultimaker2', 'ultimaker3', and other definitions but the one that I've found that doesn't work is 'fdmprinter'. 

I've attempted to use it with overrides, without overrides, and even tested it with a command. It just doesn't seem to work correctly. 

I've narrowed it down to the slicer.slice() function. And as an example so that you know that I'm using it correctly: 

```ts
 const slicer = new CuraWASM({

    definition: definition ? resolveDefinition(definition) : undefined,
    overrides: overrides ? overrides : undefined,

  });

  const res = await fetch(url);
  const stl = await res.arrayBuffer();

  const { gcode } = await slicer.slice(stl, "stl");

  const blob = new Blob([gcode], { type: "text/plain" }); // This isn't throwing the error, I've placed 
                                                                                        // a debugger right above this line and it never 
                                                                                        // caught it, and when I use a debugger above the slicer.slice()
                                                                                        // it throws an error
```

Wakeful-Cloud, 2021-06-28:
@benzend Can you post the error message?

*PS: sometimes debugging breakpoints don't work if you place them on empty lines.*

benzend, 2021-06-28:
I went through looking at how you create your "resolveDefinitions" function. I noticed that you do something special for "fdmprinter" and "fdmextruder". 

[image]

I'm guessing if I simply don't add the definition specification it should default to "fdmprinter" and "fdmextruder"? Because it worked and I'm going to test the print.

benzend, 2021-06-28:
Also here is the error that I normally get: 

[image]

benzend, 2021-06-28:
Also, I found something cool you can do, I'm not exactly what the possibilities are but I noticed you can only add one metadata key-val on the printer part.

[image]

Am I able to put more meta on there or is the limit rather than just TypeScript complaining?

Wakeful-Cloud, 2021-06-29:
> I've tried using 'ultimaker2', 'ultimaker3', and other definitions but the one that I've found that doesn't work is 'fdmprinter'. I noticed that you do something special for "fdmprinter" and "fdmextruder".

Correct, because every printer definition relies on `fdmprinter` and/or `fdmextruder`, these are shipped with cura-wasm itself while other definitions are shipped in `cura-wasm-definitions`. You can see [here](https://github.com/Cloud-CNC/cura-wasm/blob/master/src/worker.ts#L72) that `fdmprinter` and `fdmextruder` (Primary definitions) are loaded into the virtual filesystem separately from each other **and** everything else (As opposed to resolved/secondary definitions which are combined into a single JSON document and loaded as a single file). If you examine how the [CLI arguments are generated](https://github.com/Cloud-CNC/cura-wasm/blob/master/src/arguments.ts#L30), you'll see that Cura WASM expects the secondary definition to be located at `definitions/printer.def.json`. So when you use either `fdmprinter` or `fdmextruder` (Which resolve to `undefined` as you saw), an empty file is created at `definitions/printer.def.json` (Instead of a secondary definition). This likely tricks Cura Engine into slicing the STL but yielding bad GCODE (Probably why your printer didn't move). Having the definition split between two repositories is going to change in [V2](https://github.com/Cloud-CNC/cura-wasm/discussions/11), but in the meantime as long as you set the definition file (`-j` argument) to `definitions/fdmprinter.def.json` or `definitions/fdmextruder.def.json` (Via the `command` option), you should be fine.

> Hopefully they're in the right format, as in TS, the values are only accepted as strings.

This is because all options are [converted to CLI arguments](https://github.com/Cloud-CNC/cura-wasm/blob/master/src/arguments.ts#L14) (Which don't have data types).

> Am I able to put more meta on there or is the limit rather than just TypeScript complaining?

You're able to define arbitrary data (Which will be stringified); only the [fields necessary for Cura WASM internals are defined](https://github.com/Cloud-CNC/cura-wasm-definitions/blob/master/src/types.ts#L45).

If you're still getting an error, please post more than just the promise rejection message (Even if heavily redacted), thanks!

Wakeful-Cloud, 2021-06-29:
Yes, that command should work. Thanks for your interest in Cura WASM!

closed 2021-06-29 by Wakeful-Cloud