Cloud-CNC/cura-wasm #15: Speed setting not taking effect

JayEnaR, 2021-12-18:
Hi, thanks for the time and effort in creating this library.
When I pass in the parameter to change the speed, there is no difference in the print time output.

command:
`'slice -j definitions/fdmprinter.def.json -o Model.gcode -s speed_print=120 -s layer_height=0.2 -l Model.stl'`

Is this a bug, or am I doing something wrong?
Also, how do I pass in a custom def.json?

Wakeful-Cloud, 2021-12-18:
I'm sorry to hear you're having difficulties.

> When I pass in the parameter to change the speed, there is no difference in the print time output

Your command looks correct, but have you tried running it with the [native Cura Engine](https://github.com/ultimaker/curaengine) **with the same definitions/printer**?

> using the `resolveDefinition` also does not seem to work when using overrides either

Try omitting the scope option or set it to `e0`.

If neither of the above work, try enabling `verbose` logging and see what command Cura Engine is invoked with.

> `[r]esolveDefinition('fdmprinter')` returns null

`fdmprinter` and `fdmextruder` are special. Because virtually all printers rely on them, they're bundled into `cura-wasm` - not `cura-wasm-definitions`

> Also, how do I pass in a custom def.json?

This isn't really a supported feature in V1, but should be easy to do in V2 (Still under development). That said, you can pass any value that satisfies the [`CombinedDefinition` type](https://github.com/Cloud-CNC/cura-wasm-definitions/blob/1c17d573a07ef91aea8a28e14a088206ca178891/src/types.ts#L45) to the `definition` property in the Cura WASM constructor and it will get stringified and written to the filesystem (See [`worker.ts:85`](https://github.com/Cloud-CNC/cura-wasm/blob/09c61a5c0519991eafa21b39c44e68c4715d9f09/src/worker.ts#L85)). You could probably even get away with the following (Though I haven't tested it):
```json
{
  "extruders": [
    "arbitrary, definitely not JSON content" //Written to /definitions/extruder-0.def.json
    "more non-JSON content" //Written to /definitions/extruder-1.def.json
  ],
  "printer": "Binary data: \u0048\u0065\u006c\u006c\u006f\u002c\u0020\u0077\u006f\u0072\u006c\u0064\u0021" //Written to /definitions/printer.def.json
}
```

The important thing is that all definitions (Excluding `fdmprinter` and `fdmextruder`) are bundled into one (See [`cura-wasm-definitions/index.ts:93`](https://github.com/Cloud-CNC/cura-wasm-definitions/blob/1c17d573a07ef91aea8a28e14a088206ca178891/src/index.ts#L93) for the algorithm).

JayEnaR, 2021-12-25:
Sorry for the late reply.
It seems the order of the command is correct. So, what worked was using the `fdmpinter.def.json` but making sure that the -s commands used have no child elements.

closed 2022-01-29 by Wakeful-Cloud