Cloud-CNC/cura-wasm #16: Cannot set settings

EZKOWW, 2022-04-27:
Hi!

I am currently implementing Cura-WASM (installed through NPM) in a Node JS web application, but simply cannot figure out why my parameters aren't taken into account while slicing. My main goal is to estimate an object's price based on its volume and the infill required by the customer. However, whenever I try to implement "infill_sparse_density=X", or any other setting, it doesn't seem to affect the output print time whatsoever. I have looked everywhere, but couldn't find anything on this issue.

Here's the code I currently have for slicing, let me know if anything else is needed

    const slicer = new CuraWASM({
    
         command: 'slice -v -j definitions/fdmprinter.def.json -o Model.gcode -s layer_height=0.2 -s infill_sparse_density=50 -l Model.stl',

         verbose: true,
    });

    const res = await file;
    const stl = await res.arrayBuffer();

    slicer.on('progress', percent => {
      //...
    });

    const gcodeFile = await slicer.slice(stl, 'stl');
    }
`
Thanks for your time,
Really awesome software by the way!

Wakeful-Cloud, 2022-04-27:
Would you mind posting the output from Cura WASM (Especially the stuff generated by the verbose setting) and the Cura WASM version? This could be related to #8 and #15.

EZKOWW, 2022-04-27:
Version is "cura-wasm 1.5.2". Doesn't seem like anything crazy is going on in the logs though, apart from the layer_height at the end visibly not changing. Here's everything : 

[image]
[image]
[image]
[image]

Thank you for the insane response speed!

Wakeful-Cloud, 2022-04-27:
I agree - those logs look pretty normal. Can you try searching for `layer_height` or `infill_sparse_density` in the console? Also, are you able to run `CuraEngine slice -v -j definitions/fdmprinter.def.json -o Model.gcode -s layer_height=0.2 -s infill_sparse_density=50 -l Model.stl` with desktop Cura? That will help determine if this is related to Cura Engine itself, or isolated to just Cura WASM.

EZKOWW, 2022-04-28:
Sure! 

And sorry for the delay.

Aside from the layer height at the end, this is what the logs show. Seems like the "-s layer_height" parameter isn't being affected by the command.

[image]
[image]
[image]

And here is CuraEngine's output in the console. Modifying the "-s infill_sparse_density" parameter still changes nothing.
Could it be related to "-s support_infill_extruder_nr" being set to the wrong extruder number?

This issue also can't possibly be related to my computer, since it happens on every device I host the Node JS application on. Perhaps it could also have to do with the way I installed it? 

I'm just guessing at this point, but now that it seems to also be a problem with CuraEngine (possibly only my installation?), I'll do more research on my end and try to figure some stuff out. 

```
Cura_SteamEngine version 4.7.0
Copyright (C) 2020 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
[WARNING] Unrecognized data type in JSON setting machine_disallowed_areas
[WARNING] Unrecognized data type in JSON setting nozzle_disallowed_areas
[WARNING] Unrecognized data type in JSON setting machine_head_with_fans_polygon
loading 'Holder.STL' took 0.015 seconds
Loaded from disk in 0.031s
[WARNING]  -s mesh_rotation_matrix="[[1,0,0], [0,1,0], [0,0,1]]" -s mesh_position_y="0" -s center_object="false" -s small_feature_speed_factor_0="50" -s small_feature_speed_factor="50" -s small_feature_max_length="0" -s wipe_hop_speed="10" -s wipe_hop_amount="1" -s wipe_hop_enable="true" -s wipe_pause="0" -s wipe_retraction_retract_speed="3" -s wipe_retraction_amount="1" -s max_extrusion_before_wipe="10" -s clean_between_layers="false" -s wipe_retraction_extra_prime_amount="0" -s bridge_skin_density_3="80" -s bridge_skin_material_flow_3="110" -s bridge_skin_speed_3="15" -s bridge_fan_speed_2="0" -s bridge_skin_density_2="75" -s bridge_skin_speed_2="25" -s bridge_enable_more_layers="true" -s bridge_skin_density="100" -s bridge_wall_material_flow="50" -s bridge_sparse_infill_max_density="0" -s bridge_skin_support_threshold="50" -s wall_overhang_angle="90" -s adaptive_layer_height_variation_step="0.01" -s adaptive_layer_height_enabled="false" -s wireframe_nozzle_clearance="1" -s wireframe_roof_outer_delay="0.2" -s wireframe_roof_fall_down="2" -s wireframe_straight_before_down="20" -s wireframe_strategy="compensate" -s wireframe_drag_along="0.6" -s wireframe_fall_down="0.5" -s wireframe_top_jump="0.6" -s wireframe_up_half_speed="0.3" -s wireframe_bottom_delay="0" -s wireframe_top_delay="0" -s wireframe_printspeed_flat="5" -s wireframe_printspeed_down="5" -s wireframe_printspeed_up="5" -s wireframe_printspeed_bottom="5" -s adaptive_layer_height_variation="0.1" -s flow_rate_extrusion_offset_factor="100" -s flow_rate_max_extrusion_offset="0" -s wipe_retraction_enable="true" -s magic_fuzzy_skin_point_dist="0.8" -s magic_fuzzy_skin_thickness="0.3" -s magic_fuzzy_skin_outside_only="false" -s wireframe_flat_delay="0.1" -s magic_fuzzy_skin_enabled="false" -s support_conical_angle="30" -s support_conical_enabled="false" -s cross_infill_density_image="" -s cross_infill_pocket_size="2" -s coasting_speed="90" -s conical_overhang_enabled="false" -s draft_shield_height="10" -s draft_shield_height_limitation="full" -s draft_shield_dist="10" -s draft_shield_enabled="false" -s wipe_repeat_count="5" -s bridge_fan_speed_3="0" -s support_skip_some_zags="false" -s minimum_polygon_circumference="1" -s material_flow_dependent_temperature="false" -s roofing_pattern="lines" -s slicing_tolerance="middle" -s smooth_spiralized_contours="true" -s coasting_min_volume="0.8" -s magic_mesh_surface_mode="normal" -s mold_angle="40" -s mold_roof_height="0.5" -s mold_width="5" -s mold_enabled="false" -s cutting_mesh="false" -s meshfix_maximum_travel_resolution="1" -s meshfix_maximum_resolution="0.5" -s remove_empty_first_layers="true" -s alternate_carve_order="true" -s switch_extruder_extra_prime_amount="0" -s switch_extruder_prime_speed="20" -s cross_support_density_image="" -s switch_extruder_retraction_amount="20" -s wireframe_height="3" -s ooze_shield_dist="2" -s ooze_shield_angle="60" -s prime_tower_position_y="200" -s prime_tower_min_volume="6" -s prime_tower_size="20" -s prime_tower_enable="false" -s raft_base_fan_speed="0" -s raft_interface_fan_speed="0" -s raft_surface_fan_speed="0" -s raft_base_jerk="20" -s raft_interface_jerk="20" -s wall_overhang_speed_factor="100" -s raft_base_acceleration="3000" -s raft_interface_acceleration="3000" -s raft_base_speed="15" -s raft_surface_speed="20" -s raft_base_line_spacing="1.6" -s raft_base_line_width="0.8" -s raft_interface_line_spacing="0.9" -s raft_interface_thickness="0.15" -s raft_surface_line_width="0.4" -s raft_surface_thickness="0.1" -s material_flush_purge_speed="0.5" -s retraction_hop_after_extruder_switch="true" -s material_break_preparation_speed="2" -s material_break_preparation_retracted_position="-16" -s m [... cut ...]
Progress: support accomplished in 0.110s
Starting export...
Multithreading GcodeLayerThreader with 12 threads.
Progress: export accomplished in 1.187s
Total time elapsed  1.42s.
Gcode header after slicing:
;FLAVOR:Marlin
;TIME:58555
;Filament used: 31.5552m
;Layer height: 0.2
;MINX:43.546
;MINY:62.2
;MINZ:0.3
;MAXX:255.361
;MAXY:215.835
;MAXZ:39.9
End of gcode header.
Print time (s): 58555
Print time (hr|min|s): 16h 15m 55s
Filament (mm^3): 201303
```

Thank you so much for your support through this, it really is appreciated!

EZKOWW, 2022-04-28:
Alright!

I'll do further research on this,

Thanks a lot for the support!